bblanchon/ArduinoStreamUtils #1: Debug logging

Daemach, 2019-05-24:
I'm working on an ESP32 project now where the devices must log offline data locally (to SPIFFS and/or SD) as well as debugging information.  My solution is not elegant, but it does work.  I have a debug function that you can pass a string to and it will send the results to both the serial port and the SD card.  It works, but it's clumsy and not robust. I also have uploads working, so debug files get posted as a multipart form to a web server automatically daily or whenever wifi becomes available.  My methods aren't elegant but they seem to be reliable. 

I've always wanted something that can take a printf statement and send the data to serial, SD, SPIFFS and/or a webserver (remote logging).  Even better would be the ability to pass it a "debug level" info, warning, error, etc., and control the level of logging by passing it a flag from the server in a json config file.  

I don't know if you're interested in helping with something like this, but I'll happily offer any and all of my working code to use as a starting point if you can.  Here's what I would like to see:

```
typedef  struct {
    bool bSerial = true;
    bool bSD = true;
    bool bSPIFFS = false;
    bool bWebServer = true;
    bool bSSL = false;
    char *username = "username";
    char *password = "password";
    char *fingerprint = ""; //for ESP8266 or other constrained devices that can't do full SSL
    bool bRTC = true;  
    char *fileTemplate = "/debug/{d}.log";  // {date - yyyymmdd}  I use a realtime clock, but timeLib works too, or SNTP.  Anything with a now() function
    char *url = "debug.mydomain.com/logme.php";
    char *formVariable = "data";
    byte *port = 80;
    byte debugLevel = 0;  //0 = none, 1 = error, 2 = error + warning, 3 = error + warning + info, 4 = lvl 3 + debugging info
    long version = 0;  //this could be populated as a config file download using arduinoJson  If the server version is different than the local version , download new config, save to SPIFFS reboot and reparse
} debugLog

debug.printf(3, "LightLevel  = %d, humidity = %02f\n", lightsOn, humidity); //first param is debugging level 3 - info
debug.printf(2, "Wifi disconnected\n"); //first param is debugging level 2 - warning
debug.printf(1, "Some ugly error %d, %d, %02f\n", param1, param2, param3); //first param is debugging level 1 - error  
```

bblanchon, 2019-05-28:
Hi @Daemach,

I'm pretty sure there other libraries to handle log with level and format strings.
Have you checked?

With ArduinoTrace and StreamUtils, I took a different approach where logging is only used as a debugging tool and removed before production.

Best Regards,
Benoit

Daemach, 2019-05-28:
I haven't been able to find anything that does everything that I need.  I guess I'll keep looking.

closed 2019-05-28 by Daemach