bblanchon/ArduinoStreamUtils #2: Errors on latest version when using esp8266

ErrorErrorError, 2019-06-01:
Hello, 

I recently learned about this library after using ArduinoJSON and when I use this library and compile, I get this error `In file included from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/LoggingClient.hpp:7:0,
from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils.h:5,
from src/main.cpp:6:
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/ConnectForwardingPolicy.hpp: In member function 'bool StreamUtils::ConnectForwardingPolicy::stop(Client&, unsigned int)':
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/ConnectForwardingPolicy.hpp:28:31: error: no matching function for call to 'Client::stop(unsigned int&)'
return target.stop(timeout);
^
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/ConnectForwardingPolicy.hpp:28:31: note: candidate is:
In file included from /Users/erikb/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClient.h:27:0,
from /Users/erikb/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from src/main.cpp:4:
/Users/erikb/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:38:22: note: virtual void Client::stop()
virtual void stop() = 0;
^
/Users/erikb/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:38:22: note:   candidate expects 0 arguments, 1 provided
In file included from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/LoggingClient.hpp:9:0,
from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils.h:5,
from src/main.cpp:6:
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/WriteLoggingPolicy.hpp: In member function 'bool StreamUtils::WriteLoggingPolicy::flush(Client&, unsigned int)':
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/WriteLoggingPolicy.hpp:34:32: error: no matching function for call to 'Client::flush(unsigned int&)'
return target.flush(timeout);
^
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/WriteLoggingPolicy.hpp:34:32: note: candidate is:
In file included from /Users/erikb/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClient.h:27:0,
from /Users/erikb/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from src/main.cpp:4:
/Users/erikb/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:37:22: note: virtual void Client::flush()
virtual void flush() = 0;
^
/Users/erikb/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:37:22: note:   candidate expects 0 arguments, 1 provided
In file included from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/LoggingClient.hpp:10:0,
from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils.h:5,
from src/main.cpp:6:
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteLoggingPolicy, StreamUtils::ConnectForwardingPolicy>':
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/LoggingClient.hpp:14:24:   required from here
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ClientProxy.hpp:69:7: error: 'int StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::connect(const IPAddress&, uint16_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; uint16_t = short unsigned int]' marked override, but does not override
int connect(const IPAddress &ip, uint16_t port) override {
^
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ClientProxy.hpp:85:8: error: 'bool StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::stop(unsigned int) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy]' marked override, but does not override
bool stop(unsigned timeout = 0) override {
^
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ClientProxy.hpp:105:8: error: 'bool StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::flush(unsigned int) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy]' marked override, but does not override
bool flush(unsigned timeout = 0) override {
^
In file included from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ReadBufferingClient.hpp:9:0,
from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils.h:6,
from src/main.cpp:6:
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/WriteForwardingPolicy.hpp: In member function 'bool StreamUtils::WriteForwardingPolicy::flush(Client&, unsigned int)':
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/WriteForwardingPolicy.hpp:26:32: error: no matching function for call to 'Client::flush(unsigned int&)'
return client.flush(timeout);
^
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/../Policies/WriteForwardingPolicy.hpp:26:32: note: candidate is:
In file included from /Users/erikb/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClient.h:27:0,
from /Users/erikb/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from src/main.cpp:4:
/Users/erikb/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:37:22: note: virtual void Client::flush()
virtual void flush() = 0;
^
/Users/erikb/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Client.h:37:22: note:   candidate expects 0 arguments, 1 provided
In file included from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/LoggingClient.hpp:10:0,
from /Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils.h:5,
from src/main.cpp:6:
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteForwardingPolicy, StreamUtils::ConnectForwardingPolicy>':
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ReadLoggingClient.hpp:14:28:   required from here
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ClientProxy.hpp:69:7: error: 'int StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::connect(const IPAddress&, uint16_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteForwardingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; uint16_t = short unsigned int]' marked override, but does not override
int connect(const IPAddress &ip, uint16_t port) override {
^
/Users/erikb/.platformio/lib/StreamUtils_ID6347/src/StreamUtils/Clients/ClientProxy.hpp:85:8: error: 'bool StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::stop(unsigned int) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteForwardingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy]' marked override, but does not override
[... 95 lines cut ...]
            } else if(t.equals("DeviceSolid")){
                solid(doc);
            } else if(t.equals("DeviceWaves")){
                waves(doc);
            }
        }
        Serial.println(" ");
        client.flush();
        client.stop();
        delay(10);
    }
}

void waves(DynamicJsonDocument document) {

}

void solid(DynamicJsonDocument doc){
    brightness = doc["brightness"].as<int8_t>();

}

void music(DynamicJsonDocument doc){

}`

bblanchon, 2019-06-03:
Hi @ErrorErrorError,

Thank you for reporting this.
It seems that the team reverted the quirks they inserted in the ESP8266 core:
https://github.com/esp8266/Arduino/commit/cdb549572d25cb1abf0658ff285009d591db1f3a#diff-6b1b07c053c3e5f501aa2b480d825226
That's cool, I can remove my workarounds then.

Meanwhile, you can disable the workarounds with these settings:

```c++
#define STREAMUTILS_CLIENT_CONNECT_TAKE_CONST_REF 0
#define STREAMUTILS_CLIENT_FLUSH_TAKES_TIMEOUT 0
#define STREAMUTILS_CLIENT_STOP_TAKES_TIMEOUT 0
#include <StreamUtils.h>
```

Best Regards,
Benoit

ErrorErrorError, 2019-06-03:
@bblanchon Thank you very much!

closed 2019-06-05 by bblanchon

AngeloEGMdeSousa, 2022-09-01:
Hello @bblanchon,
I had this similar error:

In file included from /mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/HammingClient.hpp:11:0,

from /mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils.hpp:5,

from /mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils.h:5,

from /tmp/652260870/Projeto_Rega_IoT_aug31a/Projeto_Rega_IoT_aug31a.ino:3:

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteLoggingPolicy, StreamUtils::ConnectForwardingPolicy>':

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/LoggingClient.hpp:14:24: required from here

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:76:7: error: 'int StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::connect(IPAddress, uint16_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; uint16_t = short unsigned int]' marked override, but does not override

int connect(IPAddress ip, uint16_t port) override {

^

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:88:8: error: 'void StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::stop() [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy]' marked override, but does not override

void stop() override {

^

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteForwardingPolicy, StreamUtils::ConnectForwardingPolicy>':

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ReadLoggingClient.hpp:14:28: required from here

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:76:7: error: 'int StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::connect(IPAddress, uint16_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteForwardingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; uint16_t = short unsigned int]' marked override, but does not override

int connect(IPAddress ip, uint16_t port) override {

^

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:88:8: error: 'void StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::stop() [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteForwardingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy]' marked override, but does not override

void stop() override {

^

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadForwardingPolicy, StreamUtils::WriteLoggingPolicy, StreamUtils::ConnectForwardingPolicy>':

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/WriteLoggingClient.hpp:15:7: required from here

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:76:7: error: 'int StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::connect(IPAddress, uint16_t) [with ReadPolicy = StreamUtils::ReadForwardingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; uint16_t = short unsigned int]' marked override, but does not override

int connect(IPAddress ip, uint16_t port) override {

^

/mnt/create-efs/webide/1c/e2/1ce2f752a26da7808c22fbd4e6f3fd50:bluemot1on/libraries_v2/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:88:8: error: 'void StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::stop() [with ReadPolicy = StreamUtils::ReadForwardingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy]' marked override, but does not override

void stop() override {

^

Error during build: exit status 1

I'm using NodeMCU with this code:

// StreamUtils - Version: 1.6.3
#include <StreamUtils.h>
#include <StreamUtils.hpp>
#define STREAMUTILS_CLIENT_CONNECT_TAKE_CONST_REF 0
#define STREAMUTILS_CLIENT_FLUSH_TAKES_TIMEOUT 0
#define STREAMUTILS_CLIENT_STOP_TAKES_TIMEOUT 0
#define ARDUINOJSON_ENABLE_PROGMEM 0
[... 177 lines cut ...]

      DeserializationError error = deserializeJson(comandosok, mySerial); 
       if (error) {
       Serial.print("deserializeJson() failed with code ");
       Serial.println(error.c_str());
       return;
       }

       ACK1 = comandosok["ACK1"];

      if(ACK1==1){state=0;}
    }
    break;
  }
}

void loop() {
  ArduinoCloud.update();
  
  MasterTxRx();
}

What could be the problem?

Thank you.

bblanchon, 2022-09-02:
@AngeloEGMdeSousa, I just tested with [ESP8266 core](https://github.com/esp8266/Arduino) version 3.0.2, and it worked fine.

Are you using another core?
If so, [**open a new issue**](https://github.com/bblanchon/ArduinoStreamUtils/issues/new/choose) and include the core's name and version.

commit eb2801e0 (2019-06-05): Fix compatibility with ESP8266 core 2.5.1+ (closes #2)