bblanchon/ArduinoStreamUtils #5: json, sd write performance

salvq, 2020-01-08:
I am about to see if get better sd card write performance with recommend approach json write to sd card

**This is what I read from original post**
```
File file = SD.open(filename, FILE_WRITE);
WriteBufferingStream bufferedFile(client, 64);
doc["hello"] = "world";
serializeJson(doc, loggedFile);
bufferedFile.flush();
file.close();
```

**Should not example have describe like this rather that above one**
```
File file = SD.open(filename, FILE_WRITE);
WriteBufferingStream bufferedFile(file, 64);
doc["hello"] = "world";
serializeJson(doc, bufferedFile);
bufferedFile.flush();
file.close();
```

Is there any recommend buffer size (above example 64) ? Does it depends on size of sd card ie. cluster size or json size or...I am using `#include <SdFat.h>` library for SD card writing.

bblanchon, 2020-01-09:
Hi @salvq,

I'm sorry, I don't have any recommendation for the SD library.
You should ask the authors of the library; they probably ran some tests.
BTW, I'd be very interested in knowing the answer...

Best regards,
Benoit

salvq, 2020-01-30:
Based on what I found on library and forums seem that SdFat has a single 512 byte buffer for SD I/O so expecting this is the optimal value

closed 2020-01-30 by salvq

bblanchon, 2020-01-30:
If they already implement an internal buffer, it may be counter-productive to use `WriteBufferingStream`.
If I were you, I would try with and without, and compare the results.