bblanchon/ArduinoStreamUtils #8: Error trying ot log while deserializing

weshouman, 2021-01-20:
When trying to follow this [example](https://arduinojson.org/v6/how-to/debug-data-coming-from-a-stream/) to log a string while deserializing it I get either of the following errors (when using either a `String` or `char *` input)  

`error: no matching function for call to 'StreamUtils::ReadLoggingStream::ReadLoggingStream(String&, HardwareSerial&)`  
`error: no matching function for call to 'StreamUtils::ReadLoggingStream::ReadLoggingStream(char*&, HardwareSerial&)`  

Sample code  
```cpp
  String text = extractContent(*filename);
  // char *textchar = extractContentChar(*filename);

  ReadLoggingStream loggingStream(text, Serial);
  // ReadLoggingStream loggingStream(textChar, Serial);
  DeserializationError error = deserializeJson(ds, loggingStream);
```

bblanchon, 2021-01-20:
Hi @weshouman,

Indeed this library, StreamUtils, only works with streams.
Your program doesn't use streams but strings.

If you want to see the content of the string, simply call `Serial.print()`:

```c++
String text = extractContent(*filename);
Serial.println(text);
DeserializationError error = deserializeJson(ds, text);
```

BTW, as a best practice, I recommend that you let `deserializeJson()` read from the `File` directly.
See [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/) for an example.

Lastly, if for some reason, you absolutely need to build a `ReadLoggingStream` on top of a `String`, you need to create a `StringStream` first:

```c++
StringStream stream = extractContent(*filename);
ReadLoggingStream loggingStream(stream, Serial);
DeserializationError error = deserializeJson(ds, loggingStream);
```

Best regards,
Benoit

weshouman, 2021-01-20:
Thanks a lot ^_^

closed 2021-01-20 by weshouman