bblanchon/ArduinoStreamUtils #14: Always flush after...

dpetican, 2021-08-17:
You do mention this as optional vis-a-vis the destructor calls flush.  If the buffer is global and you send less than the buffer size and you want to see the ouptut right away then you have to flush.  Is this correct?

bblanchon, 2021-08-18:
Hi @dpetican,

Indeed, if the buffer is global, its destructor never runs, so you must call `flush()` explicitly. 

Beet Regards,
Benoit

dpetican, 2021-08-18:
I suggest to make this explicit in the documentation.  Thanks.

bblanchon, 2021-08-19:
Sure. How would you phrase it?

dpetican, 2021-08-20:
After this sentance:

> Calling flush() is recommended but not mandatory. If you don't call it, the destructor of WriteBufferingStream will do it for you.

When WriteBufferingStream is declared global, calling flush() is mandatory.  However, you can keep adding to the buffer until you want the write to actually happen.

I didn't add: since there is no destructor for global variables.  That would be pedantic.

commit 18770f0d (2021-08-20): Don't say that calling `flush()` is optional (fixes #14)
closed 2021-08-20 by bblanchon