bblanchon/ArduinoStreamUtils #15: Optimal buffer size for hardware serial

dpetican, 2021-08-17:
Thanks for this great library(s).  I've been using ArdionJson v5 for several years and am upgrading the project and moving to v6.  One of the issues I've had is garbled JSON because of sending large messages (up to 512 bytes) at 9600 baud with minimal time in-between.  In addition to StreamUtils I will also be using message pack, hence one of the reasons for upgrading ArduinoJson.

Without (then) knowing that the parser processes input byte-by-byte it seemed to me that I wasn't consuming the hardware buffer fast enough before the next message came in and an overrun occurred.  Inserting a delay between the messages solved the issue.  I also increased the hardware read/write buffers to 128 though I'm not sure if the delay is more important.

Given this, is there a relationship between the hardware buffer size, baudrate, and software buffer size that can be defined?  If not numerically at least in a relative sense?

I can see this is actually a flow control problem.  I wonder if its possible to implement software flow control into StreamUtils.  But then it might not work with message pack.

bblanchon, 2021-08-18:
Hi @dpetican,

From my experience, there is no benefit to adding a buffer to serial ports because they already include an internal buffer. 

Regarding `ReadThrottlingStream`, it allows reducing the reading speed. I implemented it to workaround a temporary issue, but i never published it in the public API.

Best Regards,
Benoit

dpetican, 2021-08-18:
> Regarding `ReadThrottlingStream`, it allows reducing the reading speed. I implemented it to workaround a temporary issue, but i never published it in the public API.

Well reducing the reading speed is not what I really want in the end.

> From my experience, there is no benefit to adding a buffer to serial ports because they already include an internal buffer.

I'm wondering when reading from a stream buffer does deserializeJson still read byte by byte?  Then I can see that for only increasing reading speed on serial there is no advantage to a stream buffer.

So essentially my current solution is correct.  Lower the _actual_ baud rate for large frequent messages and/or increase the hardware buffer and/or reduce the _effective_ baud rate by inserting a delay between messages.  A larger hardware buffer will also allow smaller messages to be sent faster.  And implement message pack.

bblanchon, 2021-08-19:
I’m not following. What do you want to add in the documentation?

dpetican, 2021-08-20:
> From my experience, there is no benefit to adding a buffer to serial ports because they already include an internal buffer.

That or similar.  However, should that be qualified because adding logging streams to serial could be useful?  I haven't tried that yet.

I was under the impression that the stream buffer would pull a chunk of data out of the hardware serial buffer.  This may or may not be true.  Still I think the more important point is the deserializer would still pull 1 byte at a time from the stream buffer.  And the program will block waiting on the deserializer in either scenario.  Might as well increase the hardware buffer size.  I think the point of your library is really to add a buffer where none existed before.  Buffer on top of existing buffer does nothing.  Well at least for serial with nearly blocking read/writes.  If we were talking about true multithreading then perhaps a stream buffer on serial might be useful.

As for everything to do with baud rates, flow control etc,  thats probably beyond the scope of what you are trying to accomplish with this library.

commit e645ffed (2021-08-20): Say that serial ports can't benefit from buffering (fixes #15)
closed 2021-08-20 by bblanchon