bblanchon/ArduinoStreamUtils #16: Cannot use a StringStream in ArduinoJson's deserializeJson

LeedanJohnson, 2022-02-08:
I hope I'm not doing anything wrong here, but anytime I try to use a StringStream as the source for ArduinoJson's deserializeJson method, it always results in an EmptyInput error.

`
DynamicJsonDocument doc(512);
String testString= "{\"testField\":\"testValue\"}";
StringStream testStringStream(testString);
auto jsonErr = deserializeJson(doc, testStringStream);

if (jsonErr)
{
    Serial.printf("I always get this EmpyInput error: %s\n", jsonErr.c_str());
    return;
}
`

bblanchon, 2022-02-09:
Hi @LeedanJohnson,

Thank you very much for reporting this bug.

I tested your program with various boards using the Arduino IDE 1.8.19.
Here are the results:

| Board | Core | Result |
|:---|:---|:---|
| Arduino Leonardo | AVR 1.8.3 | Ok |
| Arduino Nano Every | megaAVR 1.8.7 | Ok |
| Arduino mkr1000 | SAMD 1.8.12 | Ok |
| Adafruit Feather Huzzah | ESP8266 2.7.4 | EmptyInput |
| Adafruit Feather Huzzah | ESP8266 3.0.2 | EmptyInput |
| Adafruit ItsyBitsy M4 | Adafruit AVR 1.7.9 | Ok |
| ESP32-WROVER-B | ESP32 2.0.2 | EmptyInput |
| Teensy 3.1 | Teensyduino 1.56 | Ok |

This problem seems specific to ESP8266 and ESP32 cores.
I'll try to see why this happens.

I saw your PR, but I cannot accept any change that isn't backed by a unit test.
Don't worry, I'll take care of this.

Best regards,
Benoit

closed 2022-02-09 by bblanchon

bblanchon, 2022-02-09:
The bug is fixed in version 1.6.2.

BTW, prefer passing the `String` directly to `deserializeJson()` if you can; it would be much faster.

LeedanJohnson, 2022-02-09:
Thank you so much for addressing this so quickly! I greatly appreciate your work.

In this case it isn't an option, but thanks for the tip!

commit 53cf9dcd (2022-02-09): Fix `StringStream.readBytes()` on ESP8266, ESP32, and STM32 (fixes #16)
Closes #17.