bblanchon/ArduinoStreamUtils #20: STM32f103c8 maple mini

reggaelbowl, 2022-04-05:
Hi, StreamUtils doesn't work in stm32f103c8t6
In file included from C:\Users\Pablo\Documents\Arduino\libraries\StreamUtils\src/StreamUtils/Prints/BufferingPrint.hpp:9:0,
                 from C:\Users\Pablo\Documents\Arduino\libraries\StreamUtils\src/StreamUtils.hpp:13,
                 from C:\Users\Pablo\Documents\Arduino\libraries\StreamUtils\src/StreamUtils.h:5,
                 from C:\Users\Pablo\Documents\Arduino\libraries\StreamUtils\examples\WriteBuffer\WriteBuffer.ino:14:
C:\Users\Pablo\Documents\Arduino\libraries\StreamUtils\src/StreamUtils/Prints/PrintProxy.hpp: In instantiation of 'class StreamUtils::PrintProxy<StreamUtils::WriteLoggingPolicy>':
C:\Users\Pablo\Documents\Arduino\libraries\StreamUtils\src/StreamUtils/Prints/LoggingPrint.hpp:12:23:   required from here
C:\Users\Pablo\Documents\Arduino\libraries\StreamUtils\src/StreamUtils/Prints/PrintProxy.hpp:27:10: error: 'size_t StreamUtils::PrintProxy<WritePolicy>::write(const uint8_t*, size_t) [with WritePolicy = StreamUtils::WriteLoggingPolicy; size_t = unsigned int; uint8_t = unsigned char]' marked override, but does not override
   size_t write(const uint8_t *buffer, size_t size) override {
and more errors

bblanchon, 2022-04-05:
Hi @reggaelbowl,

Thank you for reporting this issue.

I just tested with [stm32duino](https://github.com/stm32duino/Arduino_Core_STM32) 2.2.0 and it compiles fine.

@reggaelbowl, which Arduino Core are you using?

Best regards,
Benoit

reggaelbowl, 2022-04-11:
hi, i see...., i am using a different stm32duino-core http://dan.drown.org/stm32duino/package_STM32duino_index.json

thanks for your response,
Pablo

bblanchon, 2022-04-12:
Hi Pablo,

I updated the library to support this core as well.

I don't have the hardware to validate that it works, so can you test the library?
To do that: first, uninstall the library, then clone this repo in your `libraries/` folder.

I'm waiting for your feedback before releasing a new version.

Best regards,
Benoit

closed 2022-04-12 by bblanchon

bblanchon, 2022-04-14:
I got my hands on an STM32F103C8 and confirmed that the library works fine.
For some reason, I could not get any `Serial.print()` to work from the `setup()` function, but it works in the `loop()` function.

commit 82652f5f (2022-04-12): Add support for Roger's core for STM32 (fixes #20)