bblanchon/ArduinoStreamUtils #24: Feature request: Calculate statistics on a (wrapped) stream

vshymanskyy, 2022-06-22:
Could be interesting to be able to grab some stats of the stream usage:
- Total TX/RX bytes count
- Total TX/RX packets count (writes)
- Min/Max/Average read/write size

The only difficulty I see here: Some functions (i.e. `readBytes`) could be implemented via `read()` on some streams, but it can also be implemented directly by other implementations. Not sure if there's a clean way to handle it.

vshymanskyy, 2022-06-24:
Sounds cool. But some names may be ambiguous esp. `readBytes()`. Maybe prefix all of them with `total`, `min`, `max`, `avg` ?

bblanchon, 2022-06-26:
You're right!
Maybe something like this, then:

| read | write |
|:---|:---|
| `readCount()` | `writeCount()` |
| `avgReadSize()` | `avgWriteSize()` |
| `maxReadSize()` | `maxWriteSize()` |
| `minReadSize()` | `minWriteSize()` |
| `totalReadSize()` | `totalWriteSize()` |

bblanchon, 2022-06-27:
I'm not sure how to handle error cases: should we count them, or ignore them?

For example, if upstream's `write()` function returns `0`, should I increment `writeCount`? should I set `minWriteSize` to `0`?

dpharris, 2022-10-23:
Maybe readErrors() and writeErrors()?

mhaberler, 2023-05-14:
for such jobs I like the header-only stats classes by https://www.johndcook.com/blog/skewness_kurtosis/

he has several such classes on his website, real collectors items - I collect them here plus my adaptations: https://github.com/mhaberler/Stats