bblanchon/ArduinoStreamUtils #29: Hammingstream using ESP32 only receiving question marks.

Klon64, 2023-02-21:
Hi Benoit,

I'm using the provided "HammingSerial1" Sketch with an ESP32 looping the TX and RX pins for testing, but only receiving "????" as outputs on the serial monitor.
So far, I've tried remapping the UART to other Pins and connecting the ESP32 to an Arduino Nano Every (with level shifter), which resulted in further question marks on the ESP side, despite the Nano working flawlessly.
ESP32 Board: [https://www.az-delivery.de/en/products/esp32-developmentboard](https://www.az-delivery.de/en/products/esp32-developmentboard)

Best regards
Felix

bblanchon, 2023-02-22:
Hi @Klon64,

Can you double-check that the transmission works correctly without the `HammingStream`?
Can you share a short repro code?

Best regards,
Benoit

Klon64, 2023-02-23:
Hi Benoit,

The transmission without `HammingStream` works as intented.
```c++
#include <StreamUtils.h>

HammingStream<7, 4> eccSerial1(Serial1);

void setup() {
  Serial.begin(115200);
  delay(500);
  while (!Serial)
    continue;

  Serial1.begin(9600);
  delay(500);
  while (!Serial1)
    continue;

  // Discard any remaining data in the input buffer
  while (Serial1.available())
    Serial1.read();
}

void loop() {
  //noecc();
  // Did we receive something?
  if (eccSerial1.available())
    // Print it
    Serial.write(eccSerial1.read());

  // Do we have something to send?
  if (Serial.available())
    // Sent it
    eccSerial1.write(Serial.read());
}

void noecc() {
  // Did we receive something?
  if (Serial1.available())
    // Print it
    Serial.write(Serial1.read());

  // Do we have something to send?
  if (Serial.available())
    // Sent it
    Serial1.write(Serial.read());
}
```

Best regards,
Felix

bblanchon, 2023-02-25:
Hi Felix,

I tried your program on an Arduino Leonardo, and it works perfectly.

However, when I run it on an ESP32, it resets when it executes `Serial1.begin(9600);`.
I don't have the time to investigate why I can't use `Serial1` on my ESP32, and it doesn't seem to be related to your problem anyway.

Best regards,
Benoit

Klon64, 2023-02-25:
Hi Benoit,

`Serial1` is used for the Flash on some ESP32 boards. I tried the same Code with `Serial2` and get the same results.

```C++
#include <StreamUtils.h>

HammingStream<7, 4> eccSerial1(Serial2);

void setup() {
  Serial.begin(115200);
  delay(500);
  while (!Serial)
    continue;

  Serial2.begin(9600);
  delay(500);
  while (!Serial2)
    continue;

  // Discard any remaining data in the input buffer
  while (Serial2.available())
    Serial2.read();
}

void loop() {
  //noecc();
  ecc();
}

void noecc() {
  // Did we receive something?
  if (Serial2.available())
    // Print it
    Serial.write(Serial2.read());

  // Do we have something to send?
  if (Serial.available())
    // Sent it
    Serial2.write(Serial.read());
}

void ecc() {
    // Did we receive something?
  if (eccSerial1.available())
    // Print it
    Serial.write(eccSerial1.read());

  // Do we have something to send?
  if (Serial.available())
    // Sent it
    eccSerial1.write(Serial.read());
}
```

Best regards,
Felix

bblanchon, 2023-02-26:
Hi Felix,

Thank you very much for these instructions!
They helped me reproduce the bug and ultimately fixed the issue: I used a `char` to store a signed value, but on ESP32, the `char` type is unsigned.

I'll release a new version once the CI build passes.

Thanks a lot for reporting this bug ❤️ 

Best regards,
Benoit

commit 5b9ffa51 (2023-02-26): Fix `HammingStream` and `HammingDecodingStream` on ESP32 (fixes #29)
closed 2023-02-26 by bblanchon