bblanchon/ArduinoStreamUtils #32: size_t readBytes() marked 'override', but does not override on ESP32

charmesal, 2023-04-26:
I'm trying to get this to work to debug another part of my code but it won't compile.

I'm using Platformio and these dependencies:
	ArduinoJson@~6.17.1
	Adafruit GFX Library@~1.6.1
	Timezone@~1.2.2
	AsyncMqttClient@~0.8.2
	ESP Async WebServer@~1.2.0
	NTPClient@~3.1.0
	PathVariableHandlers@~2.0.0
	RichHttpServer@~2.0.0
	DNSServer@~1.1.0
	Bleeper@~1.0.4
	ArduinoMSGraph@^0.2.0
	StreamUtils@^1.4.1
        AsyncTCP@~1.1.1
	BatteryLevel=https://github.com/dobrishinov/18650CL.git
	Time=https://github.com/xoseperez/Time#ecb2bb1
	WiFiManager=https://github.com/sidoh/WiFiManager#async_support
        GxEPD2=https://github.com/ZinggJM/GxEPD2#1.3.8

During compilation I get the following error:
`In file included from .pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/HammingClient.hpp:11:0,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.hpp:5,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.h:5,
                 from .pio/libdeps/esp32/ArduinoMSGraph/src/ArduinoMSGraph.cpp:14:
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteLoggingPolicy, StreamUtils::ConnectForwardingPolicy>':
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/LoggingClient.hpp:14:24:   required from here
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:69:10: error: 'size_t StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::readBytes(char*, size_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; size_t = unsigned int]' marked 'override', but does not 
override
   size_t readBytes(char *buffer, size_t size) override {
          ^
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteForwardingPolicy, StreamUtils::ConnectForwardingPolicy>':
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ReadLoggingClient.hpp:14:28:   required from here
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:69:10: error: 'size_t StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::readBytes(char*, size_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteForwardingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; size_t = unsigned int]' marked 'override', but does not override
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp: In instantiation of 'class StreamUtils::ClientProxy<StreamUtils::ReadForwardingPolicy, StreamUtils::WriteLoggingPolicy, StreamUtils::ConnectForwardingPolicy>':
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/WriteLoggingClient.hpp:15:7:   required from here
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:69:10: error: 'size_t StreamUtils::ClientProxy<ReadPolicy, WritePolicy, ConnectPolicy>::readBytes(char*, size_t) [with ReadPolicy = StreamUtils::ReadForwardingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; ConnectPolicy = StreamUtils::ConnectForwardingPolicy; size_t = unsigned int]' marked 'override', but does not override
In file included from .pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:10:0,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/HammingClient.hpp:11,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.hpp:5,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.h:5,
                 from .pio/libdeps/esp32/ArduinoMSGraph/src/ArduinoMSGraph.cpp:14:
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp: In instantiation of 'class StreamUtils::StreamProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteLoggingPolicy>':
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Streams/LoggingStream.hpp:13:24:   required from here
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp:62:10: error: 'size_t StreamUtils::StreamProxy<ReadPolicy, WritePolicy>::readBytes(char*, size_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; size_t = unsigned int]' marked 'override', but does not override
   size_t readBytes(char *buffer, size_t size) override {
          ^
In file included from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.hpp:22:0,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.h:5,
                 from .pio/libdeps/esp32/ArduinoMSGraph/src/ArduinoMSGraph.cpp:14:
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Streams/ProgmemStream.hpp:35:10: error: 'size_t StreamUtils::ProgmemStream::readBytes(char*, size_t)' marked 'override', but does not override       
   size_t readBytes(char* buffer, size_t size) override {
          ^
In file included from .pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:10:0,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/HammingClient.hpp:11,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.hpp:5,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.h:5,
                 from .pio/libdeps/esp32/ArduinoMSGraph/src/ArduinoMSGraph.cpp:14:
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp: In instantiation of 'class StreamUtils::StreamProxy<StreamUtils::ReadLoggingPolicy, StreamUtils::WriteForwardingPolicy>':
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Streams/ReadLoggingStream.hpp:14:7:   required from here
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp:62:10: error: 'size_t StreamUtils::StreamProxy<ReadPolicy, WritePolicy>::readBytes(char*, size_t) [with ReadPolicy = StreamUtils::ReadLoggingPolicy; WritePolicy = StreamUtils::WriteForwardingPolicy; size_t = unsigned int]' marked 'override', but does not override
   size_t readBytes(char *buffer, size_t size) override {
          ^
In file included from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.hpp:26:0,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.h:5,
Compiling .pio\build\esp32\lib334\GxEPD2\epd3c\GxEPD2_213c.cpp.o
                 from .pio/libdeps/esp32/ArduinoMSGraph/src/ArduinoMSGraph.cpp:14:
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Streams/StringStream.hpp:64:10: error: 'size_t StreamUtils::StringStream::readBytes(char*, size_t)' marked 'override', but does not override
   size_t readBytes(char* buffer, size_t length) override {
          ^
In file included from .pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/ClientProxy.hpp:10:0,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/HammingClient.hpp:11,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.hpp:5,
                 from .pio/libdeps/esp32/StreamUtils/src/StreamUtils.h:5,
                 from .pio/libdeps/esp32/ArduinoMSGraph/src/ArduinoMSGraph.cpp:14:
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp: In instantiation of 'class StreamUtils::StreamProxy<StreamUtils::ReadForwardingPolicy, StreamUtils::WriteLoggingPolicy>':
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Streams/WriteLoggingStream.hpp:14:7:   required from here
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp:62:10: error: 'size_t StreamUtils::StreamProxy<ReadPolicy, WritePolicy>::readBytes(char*, size_t) [with ReadPolicy = StreamUtils::ReadForwardingPolicy; WritePolicy = StreamUtils::WriteLoggingPolicy; size_t = unsigned int]' marked 'override', but does not override
   size_t readBytes(char *buffer, size_t size) override {
          ^
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp: In instantiation of 'class StreamUtils::StreamProxy<StreamUtils::ReadForwardingPolicy, StreamUtils::WriteWaitingPolicy>':
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Streams/WriteWaitingStream.hpp:14:7:   required from here
.pio/libdeps/esp32/StreamUtils/src/StreamUtils/Clients/../Streams/StreamProxy.hpp:62:10: error: 'size_t StreamUtils::StreamProxy<ReadPolicy, WritePolicy>::readBytes(char*, size_t) [with ReadPolicy = StreamUtils::ReadForwardingPolicy; WritePolicy = StreamUtils::WriteWaitingPolicy; size_t = unsigned int]' marked 'override', but does not override
*** [.pio\build\esp32\libe7b\ArduinoMSGraph\ArduinoMSGraph.cpp.o] Error 1`

I'm only using it here.
`ReadLoggingStream loggingStream(https.getStream(), Serial);
DeserializationError error = deserializeJson(responseDoc, loggingStream);`

If you need any more information let me know.

bblanchon, 2023-04-26:
Hi @charmesal,

Thank you for reporting this error.
[`Stream::readBytes()` has been `virtual` for 4 years!](https://github.com/espressif/arduino-esp32/commit/119ece2b0f96ca7d5477d446a62f210128a4b19e#diff-7dace83c16ef9bda4731eec6c99a784a0baceab758c57e77ad36744652008655R100)
Please make sure the ESP32 is up to date.

Best regards,
Benoit

charmesal, 2023-04-26:
Thank you for your quick reply.

That would explain a lot!
Sadly I'm using a legacy codebase that won't let me update the framework past 1.9.0. I'm running out of heap space and other issues after upgrading. Is there a way to work around this issue?

bblanchon, 2023-04-27:
Please try:

```c++
#define STREAMUTILS_STREAM_READBYTES_IS_VIRTUAL 0
#include <StreamUtils.h>
```

charmesal, 2023-05-01:
That seems to fix the compile issue.
But it won't print to Serial. Perhaps thats a different problem entirely, if so I'll make a new issue.
The code below provides no output to Serial. A normal Serial.print() works just fine.
`ReadLoggingStream loggingStream(https.getStream(), Serial);
DeserializationError error = deserializeJson(responseDoc, loggingStream);`

bblanchon, 2023-05-03:
Did you try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)?

charmesal, 2023-05-11:
I have not done so. I got put on another project for the time being. I will check it out once I contiue. Thanks alot for your help. I appreciate it!

closed 2023-05-11 by charmesal